apache/mynewt-newt #142: semver reporting in image list is not consistent

sterlinghughes, 2018-03-06:
create-image split-microbit 0 and create-image split-microbit 0.0.0.0 both report 0.0.0

but create-image split-microbit 0.0.0.1 reports 0.0.0.1

create-image split-microbit 1 reports 1.0.0

In my mind this should always report four zeros.

 

Jacobs-Air:chippd3 jacobrosenthal$ newt build split-microbit && newt create-image split-microbit 0 && newt load split-microbit 0
Building target targets/split-microbit
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysinit-app.c
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysinit-loader.c
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysflash.c
Archiving split-microbit-sysinit-app.a
Linking /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/app/apps/blesplit/blesplit_tmp.elf
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysinit-app.c
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysinit-loader.c
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysflash.c
Archiving split-microbit-sysinit-loader.a
Linking /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/loader/apps/bleprph/bleprph_tmp.elf
Linking /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/loader/apps/bleprph/bleprph.elf
Generating ROM elf 
Linking /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/app/apps/blesplit/blesplit.elf
Target successfully built: targets/split-microbit
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysinit-app.c
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysflash.c
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysinit-loader.c
Archiving split-microbit-sysinit-app.a
Linking /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/app/apps/blesplit/blesplit_tmp.elf
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysinit-app.c
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysinit-loader.c
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysflash.c
Archiving split-microbit-sysinit-loader.a
Linking /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/loader/apps/bleprph/bleprph_tmp.elf
Linking /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/loader/apps/bleprph/bleprph.elf
Generating ROM elf 
Linking /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/app/apps/blesplit/blesplit.elf
App image succesfully generated: /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/loader/apps/bleprph/bleprph.img
App image succesfully generated: /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/app/apps/blesplit/blesplit.img
Loading app image into slot 2
Loading loader image into slot 1
Jacobs-Air:chippd3 jacobrosenthal$ newtmgr -c nimble_bleprph image list
Images:
slot=0
version: 0.0.0
bootable: true
flags: active confirmed
hash: e5a1b2038053bccac9aa3b4c297df3871328f9d5232b566fcc27d102d938e385
slot=1
version: 0.0.0
bootable: false
flags: 
hash: 33a5c44375b3859cdbe764036ea4548cef48979ca737c0a15d99e0be6a84a7a9
Split status: matching (2)

 

Jacobs-Air:chippd3 jacobrosenthal$ newt build split-microbit && newt create-image split-microbit 0.0.0.1 && newt load split-microbit 0.0.0.1
Building target targets/split-microbit
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysinit-app.c
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysinit-loader.c
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysflash.c
Archiving split-microbit-sysinit-app.a
Linking /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/app/apps/blesplit/blesplit_tmp.elf
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysinit-app.c
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysinit-loader.c
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysflash.c
Archiving split-microbit-sysinit-loader.a
[... 106 lines cut ...]
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysinit-loader.c
Compiling bin/targets/split-microbit/generated/src/split-microbit-sysflash.c
Archiving split-microbit-sysinit-loader.a
Linking /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/loader/apps/bleprph/bleprph_tmp.elf
Linking /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/loader/apps/bleprph/bleprph.elf
Generating ROM elf 
Linking /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/app/apps/blesplit/blesplit.elf
App image succesfully generated: /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/loader/apps/bleprph/bleprph.img
App image succesfully generated: /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-microbit/app/apps/blesplit/blesplit.img
Loading app image into slot 2
Loading loader image into slot 1
Jacobs-Air:chippd3 jacobrosenthal$ newtmgr -c nimble_bleprph image list
Images:
slot=0
version: 1.0.0
bootable: true
flags: active confirmed
hash: 6ee411e06979a31b5553970075fb6b4b67a06d2001c2f27227f8939e5af153d3
slot=1
version: 1.0.0
bootable: false
flags: 
hash: 3019d4dfe4088c90e197b795f450fb33fd43a571414b25ab9119ff03b7f5e889
Split status: matching (2)
Jacobs-Air:chippd3 jacobrosenthal$

ccollins476ad, 2018-04-12:
When constructing the version string, the firmware intentionally excludes the build ID if it is 0:
```
int
imgr_ver_str(struct image_version *ver, char *dst)
{
    if (ver->iv_build_num) {
        return sprintf(dst, "%u.%u.%u.%lu",
          ver->iv_major, ver->iv_minor, ver->iv_revision,
          (unsigned long)ver->iv_build_num);
    } else {
        return sprintf(dst, "%u.%u.%u",
          ver->iv_major, ver->iv_minor, ver->iv_revision);
    }
}
```

@mkiiskila , I know it has been a while, but do you recall the reason for doing this?  Any reason it would be a bad idea to change this to always generate a uniform `#.#.#.#` string?

mkiiskila, 2018-04-13:
> On Apr 12, 2018, at 9:04 PM, ccollins476ad <notifications@github.com> wrote:
> 
> When constructing the version string, the firmware intentionally excludes the build ID if it is 0:
> 
> int
> imgr_ver_str(struct image_version *ver, char *dst)
> {
>     if (ver->iv_build_num) {
>         return sprintf(dst, "%u.%u.%u.%lu",
>           ver->iv_major, ver->iv_minor, ver->iv_revision,
>           (unsigned long)ver->iv_build_num);
>     } else {
>         return sprintf(dst, "%u.%u.%u",
>           ver->iv_major, ver->iv_minor, ver->iv_revision);
>     }
> }
> @mkiiskila <https://github.com/mkiiskila> , I know it has been a while, but do you recall the reason for doing this? Any reason it would be a bad idea to change this to always generate a uniform #.#.#.# string?
> 

I assumed that not everyone will want to use build numbers within the versions, so
I made it’s appearance here optional. Totally reversible decision.
I doubt there’s anything which’ll break if ppl want to see the ‘0’ there in the end.

Hope this helps.