apache/mynewt-newt #151: Error: Error updating when running newt install

NZSmartie, 2018-03-29:
I've just installed newt following the [Getting Started guide for Linux ](https://mynewt.apache.org/latest/newt/install/newt_linux/). 
My OS is Ubuntu 16.04 LTS on Windows 10 through WSL. 

```bash
$ cd ~/Documents/nRF/projects # ~/Documents is a symbolic link to the Windows FileSystem
$ newt new newt-test
Downloading project skeleton from apache/mynewt-blinky...
Installing skeleton in newt-test...
Project newt-test successfully created.
$ cd newt-test
$ newt install
Error: Error updating
```

The error message isn't exactly detailed so I rm'd `newt-test` and created a new project and tried to install again with the verbose (`-v`) option.
```
$ newt install -v
[apache-mynewt-core]:
Downloading repository description
Downloading repository mynewt-core (branch: master; commit: master) at https://github.com/apache/mynewt-core.git
Cloning into '/tmp/newt-repo878420741'...
remote: Counting objects: 83159, done.
remote: Compressing objects: 100% (50/50), done.
remote: Total 83159 (delta 20), reused 53 (delta 20), pack-reused 83086
Receiving objects: 100% (83159/83159), 147.09 MiB | 2.40 MiB/s, done.
Resolving deltas: 100% (51097/51097), done.
Checking out files: 100% (4541/4541), done.
Will checkout branch master
Fetching new remote branches/tags
Download successful!
Stashing local changes
Will create new branch mynewt_1_3_0_tag from tag tags/mynewt_1_3_0_tag
Will create new branch mynewt_1_3_0_tag from tag tags/mynewt_1_3_0_tag
Error: Error updating
```

Not quite sure what's causing this, i've enabled debug logging (`-l "DEBUG"`) and the issue stems from git while it's doing a stash.

```
# ...
2018/03/29 13:54:38.923 [DEBUG] /usr/bin/git stash
2018/03/29 13:55:27.680 [DEBUG] o=Saved working directory and index state WIP on master: 941eee7cd Merge pull request #970 from wes3/loralog

2018/03/29 13:55:27.682 [DEBUG] /usr/bin/git tag --list
2018/03/29 13:55:27.743 [DEBUG] o=mynewt_0_8_0_b1_tag
mynewt_0_8_0_b2_rc2_tag
mynewt_0_8_0_b2_rc3_tag
mynewt_0_8_0_b2_tag
mynewt_0_8_0_rc1_tag
mynewt_0_8_0_tag
mynewt_0_9_0_rc1_tag
mynewt_0_9_0_rc3_tag
mynewt_0_9_0_tag
mynewt_1_0_0_b1_rc1_tag
mynewt_1_0_0_b1_rc2_tag
mynewt_1_0_0_b1_tag
mynewt_1_0_0_b2_rc1_tag
mynewt_1_0_0_b2_tag
mynewt_1_0_0_rc1_tag
mynewt_1_0_0_tag
mynewt_1_1_0_rc1_tag
mynewt_1_1_0_rc2_tag
mynewt_1_1_0_tag
mynewt_1_2_0_rc1_tag
mynewt_1_2_0_tag
mynewt_1_3_0_rc1_tag
mynewt_1_3_0_rc2_tag
mynewt_1_3_0_rc3_tag
mynewt_1_3_0_tag
pre_sterly_refactor

2018/03/29 13:55:27.744 [DEBUG] /usr/bin/git show-ref --verify --quiet refs/heads/mynewt_1_3_0_tag
2018/03/29 13:55:27.799 [DEBUG] o=
2018/03/29 13:55:27.799 [DEBUG] err=exit status 1
2018/03/29 13:55:27.801 [DEBUG] /usr/bin/git checkout tags/mynewt_1_3_0_tag -b mynewt_1_3_0_tag
2018/03/29 13:55:28.745 [DEBUG] o=error: Your local changes to the following files would be overwritten by checkout:
	.gitignore
	.mailmap
	.rat-excludes
	LICENSE
	README.md
	apps/blecent/pkg.yml
	apps/blecent/src/peer.c
	apps/blecsc/README.md
	apps/blecsc/pkg.yml
	apps/blecsc/src/blecsc_sens.h
	apps/blecsc/src/gatt_svr.c
	apps/blecsc/src/main.c
	apps/blecsc/syscfg.yml
	apps/blemesh/syscfg.yml
	apps/blemesh_shell/pkg.yml
	apps/blemesh_shell/syscfg.yml
	apps/bleprph/pkg.yml
	apps/blesplit/pkg.yml
	apps/bletiny/src/bletiny.h
	apps/bletiny/src/misc.c
	apps/boot/pkg.yml
	apps/bsncent/src/peer.c
	apps/btshell/pkg.yml
	apps/btshell/src/btshell.h
	apps/btshell/src/cmd.c
	apps/btshell/src/cmd_gatt.c
	apps/btshell/src/cmd_gatt.h
	apps/btshell/src/cmd_l2cap.c
	apps/btshell/src/cmd_l2cap.h
	apps/btshell/src/misc.c
	apps/lora_app_shell/src/las_cmd.c
	apps/lora_app_shell/src/main.c
	apps/loraping/pkg.yml
	apps/loraping/src/main.c
	apps/loraping/syscfg.yml
	apps/pwm_test/pkg.yml
	apps/spitest/syscfg.yml
	boot/boot_serial/src/boot_serial.c
	boot/boot_serial/src/boot_uart.c
	boot/boot_serial/syscfg.yml
	compiler/arc/compiler.yml
	compiler/arc/pkg.yml
	compiler/arm-none-eabi-m3/compiler.yml
	compiler/arm-none-eabi-m3/pkg.yml
	compiler/arm-none-eabi-m3/syscfg.yml
	compiler/sim/compiler.yml
	docs/.gitignore
	docs/Makefile
	docs/README.rst
	docs/conf.py
	docs/doxygen.xml
	docs/index.rst
	docs/network/ble/ble_hs/ble_att.rst
	docs/network/ble/ble_hs/ble_gap.rst
	docs/network/ble/ble_hs/ble_gattc.rst
	docs/network/ble/ble_hs/ble_gatts.rst
	docs/network/ble/ble_hs/ble_hs.rst
	docs/network/ble/ble_hs/ble_hs_id.rst
	docs/network/ble/ble_hs/ble_hs_return_codes.rst
	docs/network/ble/ble_intro.rst
	docs/network/ble/ble_sec.rst
	docs/network/ble/ble_setup/ble_addr.rst
	docs/network/ble/ble_setup/ble_lp_clock.rst
	docs/network/ble/ble_setup/ble_setup_intro.rst
	docs/network/ble/ble_setup/ble_sync_cb.rst
	docs/network/ble/btshell/btshell_GAP.rst
	docs/network/ble/btshell/btshell_GATT.rst
	docs/network/ble/btshell/btshell_advdata.rst
	docs/network/ble/btshell/btshell_api.rst
	docs/network/ble/mesh/index.rst
	docs/network/ble/mesh/mesh_lightning_model.jpg
	docs/network/ble/mesh/mesh_topology.jpg
	docs/network/ble/mesh/sample.rst
	docs/os/core_os/callout/callout.rst
	docs/os/core_os/context_switch/context_switch.rst
	docs/os/core_os/cputime/os_cputime.rst
	docs/os/core_os/event_queue/event_queue.rst
	docs/os/core_os/heap/heap.rst
	docs/os/core_os/mbuf/mbuf.rst
	docs/os/core_os/memory_pool/memory_pool.rst
	docs/os/core_os/mutex/mutex.rst
	docs/os/core_os/mynewt_os.rst
	docs/os/core_os/porting/port_bsp.rst
	docs/os/core_os/porting/port_cpu.rst
	docs/os/core_os/porting/port_mcu.rst
	docs/os/core_os/porting/port_os.rst
	docs/os/core_os/sanity/sanity.rst
	docs/os/core_os/semaphore/semaphore.rst
	docs/os/core_os/task/task.rst
	docs/os/core_os/time/os_time.rst
	docs/os/modules/config/config.rst
	docs/os/modules/console/console.rst
	docs/os/modules/devmgmt/device-mgmt.png
	docs/os/modules/elua/lua_init.rst
	docs/os/modules/elua/lua_main.rst
	docs/os/modules/hal/hal.rst
	docs/os/modules/hal/hal_bsp/hal_bsp.rst
	docs/os/modules/hal/hal_flash/hal_flash.rst
	docs/os/modules/hal/hal_gpio/hal_gpio.rst
	docs/os/modules/hal/hal_i2c/hal_i2c.rst
	docs/os/modules/hal/hal_os_tick/hal_os_tick.rst
	docs/os/modules/hal/hal_spi/hal_spi.rst
	docs/os/modules/hal/hal_system/hal_sys.rst
	docs/os/modules/hal/hal_timer/hal_timer.rst
	docs/os/modules/hal/hal_uart/hal_uart.rst
	docs/os/modules/hal/hal_watchdog/hal_watchdog.rst
	docs/os/modules/imgmgr/imgmgr_module_init.rst
	docs/os/modules/imgmgr/imgr_ver_parse.rst
	docs/os/modules/imgmgr/imgr_ver_str.rst
	docs/os/modules/json/json_encode_object_entry.rst
	docs/os/modules/json/json_encode_object_finish.rst
	docs/os/modules/json/json_encode_object_key.rst
	docs/os/modules/json/json_encode_object_start.rst
	docs/os/modules/json/json_read_object.rst
	docs/os/modules/logs/logs.rst
	docs/os/modules/sensor_framework/sensor_api.rst
	docs/os/modules/sensor_framework/sensor_create.rst
	docs/os/modules/sensor_framework/sensor_driver.rst
	docs/os/modules/sensor_framework/sensor_framewor
Aborting
# ...
```

Since `~/Documents` is actually a symbolic link to the Windows filesystem where all permissions are forced to 0777, my guess is that git stash is unhappy with the file's permissions not matching the git working tree. So I attempted running `newt install` again on a new project under the Linux filesystem and it succeeded.

So my issue is, are we able to tell git to ignore the file permissions when stashing/checking out a branch?

NZSmartie, 2018-03-29:
Ha. a fix (after the initial install fails) is to run the command
```bash
$ git -C repos/apache-mynewt-core/ config core.fileMode false
```

Then `newt install` completes successfully
```bash
$ newt install
apache-mynewt-core successfully installed version 1.3.0-none
```

aditihilbert, 2018-03-29:
Perhaps a PR for a documentation update would be good?

NZSmartie, 2018-03-29:
scripts (specifically `*.sh`) files may not have the executable flag when the repo is cloned. so maybe an addtional step of `find -name \*.sh -type f -exec chmod a+x {} \;` would be needed after cloning?

andrewleech, 2019-03-05:
I just had the same problem trying to install my first project, also in Windows WSL.
The git filemode fix above didn't work for me however, after that and retrying install I got a long list of errors like
```
* Warning: Parsing pkg @apache-mynewt-nimble/nimble/transport/uart config: strconv.ParseInt: parsing "MYNEWT_VAL(BLE_TRANS_UART_SYSINIT_STAGE)": invalid syntax; ignoring package.
```
I was trying to install into one of the mounted windows folder, eg `/mnt/c/Users/...` 
Starting again in one of the wsl linux native folders like `~/` fixed the problem for me.

io-tal, 2019-04-02:
> ```
> * Warning: Parsing pkg @apache-mynewt-nimble/nimble/transport/uart config: strconv.ParseInt: parsing "MYNEWT_VAL(BLE_TRANS_UART_SYSINIT_STAGE)": invalid syntax; ignoring package.
> ```

I'm getting the same issue (parsing pkg) on debian, I'm running it on a wls linux native located at ~/my_dir/ , do you have clue why there's an error?