apache/mynewt-newt #153: Allow override of `build_profile` setting per-package.

ccollins476ad, 2018-04-02:
E.g., if the user wants to debug a the "nimble/host" package on constrained hardware, they might do the following:

1. Set `target.build_profile` to `optimized`.
2. In `net/nimble/host/pkg.yml`, set `pkg.build_profile` to `debug`.

Currently, there is no way to override the build profile in an individual package.

sterlinghughes, 2018-04-02:
Would another solution to be allowing libraries to be targets?  I'm not sure I'd like to have this at the package level, as I think it becomes very confusing when optimization level is overwritten by a library within the build system.  I'd prefer a way of making this explicit at the target level.

utzig, 2018-04-02:
Wouldn't it be possible to override `build_profile` for some packages, like:

```
target.app: "apps/slinky"
target.bsp: "hw/bsp/stm32f3discovery"
target.build_profile: optimized
target.build_profile.pkg_override.debug:
  - "net/nimble"
  - "sys/console/full"
```

Just a suggestion of syntax to make an example...

ccollins476ad, 2018-04-02:
I agree - it is better if this is specified in the target.   Thanks for preventing bad idea from being implemented :).

andrzej-kaczmarek, 2018-05-21:
I agree target is the proper place for user to override build profile, however I think it would be also good to have this in package (as per initial proposal). For example, `sys/sysview` package should be always built optimized since we want this code to execute quickly and it's unlikely someone will actually debug it. Of course, it can be then overridden in target if needed.

Note that it is not possible to simply build `sys/sysview` with `pkg.cflags` set to e.g. `-O2` since `newt` sorts all flags and with debug build profile we'll have `-O2 -Og` which effectively selects `-Og`.

commit cbca1da1 (2018-07-31): Allow build_profile to be overridden per package

This fixes #153.

The build profile can be overridden for a particular pacakge in two
ways:

* (1) Specify a "pkg.build_profile" value in the package's `pkg.yml`
  file.  E.g.,

   pkg.build_profile: debug

* (2) Specify the pacakge in the target's "target.package_profiles" map.
  E.g.,

    target.package_profiles:
        'apps/blinky': debug
        '@apache-mynewt-core/sys/log/full': debug

(1) takes priority over (2).

In addition, it is now possible to override the optimization setting
("-O<x>") in a package's `pkg.cflags`.  Prior to this [... cut]
closed 2018-07-31 by ccollins476ad