apache/mynewt-newt #171: Unable to have same named packages in two separate repos

sjanc, 2018-05-30:
newt build is not taking repository when checking for package name

eg we have two samples
apache-mynewt-nimble/apps/btshell
apache-mynewt-core/apps/btshell

and then target
[janc@ix mynewt]$ newt target show btshell
targets/btshell
    app=@apache-mynewt-core/apps/btshell
    bsp=@apache-mynewt-core/hw/bsp/nrf52840pdk

It is not possible to build this:
[janc@ix mynewt]$ newt build btshell
Building target targets/btshell
Error: Two app packages in build: apps/btshell, apps/btshell

I think newt should be able to distinguish packages with same name but from different repos. Especially now that we have nimble in separate repo (and more to come).

sjanc, 2018-05-30:
new tarret show actually has is
```
[janc@ix mynewt]$ newt target show btshell
targets/btshell
    app=@apache-mynewt-core/apps/btshell
    bsp=@apache-mynewt-core/hw/bsp/nrf52840pdk
```

markup broke it as I didn't mark it as code:)

ccollins476ad, 2018-05-30:
I am not able to replicate this.  Here is my target:
```
[ccollins@ccollins-mac:~/tmp/myproj3]$ newt target show btshell-nrf52dk
targets/btshell-nrf52dk
    app=@apache-mynewt-core/apps/btshell
    bsp=@apache-mynewt-core/hw/bsp/nrf52840pdk
    build_profile=debug
```

It builds successfully for me.  My `apache-mynewt-nimble` doesn't contain any apps, so I created an apps directory, copied `apps/btshell` there, and updated the `pkg.yml` with repo prefixes.

Is there a particular branch of the nimble repo that you are using to produce this error?

sjanc, 2018-06-04:
OK, so for apps we cannot do that (since we can have only 1 app),  for lib like it could be useful to see full path in error message but this is not critical. Feel free to close it if you feel the change is not worth it.

sdalu, 2018-06-11:
I guess I've just encountered the same kind of problem where using two repositories defining a compatibility layer for mynewt under the same package name:
~~~
 @repo1/port/mynewt
 @repo2/port/mynewt
~~~
This result in one of the port_mynewt.a being overwritten and failed linking

PR #172 (2018-05-31): Always refer to packages by their full name
This fixes #171.

Packages have two forms of "name":
    * Name()
    * FullName()

`Name()` is everything except the repo prefix ("@<repo-name>").  Full name includes the repo prefix.  The normal `Name` was used in several places, causing ambiguity when two repos contained packages with the same name.
commit c735ad9a (2018-05-31): Always refer to packages by their full name

This fixes #171.

Packages have two forms of "name":
    * Name()
    * FullName()

`Name()` is everything except the repo prefix ("@<repo-name>").  Full
name includes the repo prefix.  The normal `Name` was used in several
places, causing ambiguity when two repos contained packages with the
same name.
PR #334 (2019-09-24): Include @<repo> in bin directories and filenames
This fixes #171.

Prior to this PR, files and directories under `bin` were named according to package short names.  A pacakge's short name is everything *except* its repo name.  For example,

|     |     |
| --- | --- |
| Full name | @apache-mynewt-core/sys/log/stub |
| Short name | sys/log/stub |

If two packages had the same short name (but different repos), some of their artifacts might have the same filename.  In this case, the binary would fail to link.

After this PR, newt names artifacts based on package full names.  For example, choosing [... cut]
commit a15dea05 (2019-09-24): Include @<repo> in bin directories and filenames

This fixes #171.

Prior to commit, files and directories under `bin` were named according
to package short names.  A pacakge's short name is everything *except*
its repo name.  For example,

     Full name: @apache-mynewt-core/sys/log/stub
    Short name: sys/log/stub

If two packages had the same short name (but different repos), some of
their artifacts might have the same filename.  In this case, the binary
would fail to link.

After this commit, newt names artifacts based on package full names.
For example, choosing one `.a` file at random:
 [... cut]
closed 2019-09-24 by ccollins476ad