apache/mynewt-newt #242: MingW/Windows path issue causes newt load to fail

MarkMoss, 2018-11-18:
I'm following the Blinky tutorial on STM32F4-Discovery at https://mynewt.apache.org/latest/tutorials/blinky/blinky_stm32f4disc.html.  While following the tutorial on MinGW/Windows, the newt load command (`"newt load stm32f4disc_boot"`) fails as shown below:

```$ newt load -v stm32f4disc_boot
Loading bootloader
Load command: C:/msys64/home/mmoss/dev/myproj/repos/apache-mynewt-core/hw/bsp/stm32f4discovery/stm32f4discovery_download.cmd C:/msys64/home/mmoss/dev/myproj/repos/apache-mynewt-core/hw/bsp/stm32f4discovery bin\targets\stm32f4disc_boot\app\apps\boot\boot
Environment:
* BOOT_LOADER=1
* FEATURES=BASELIBC_PRESENT BOOT_LOADER CONSOLE_UART_BAUD CONSOLE_UART_DEV CONSOLE_UART_FLOW_CONTROL FLASH_MAP_MAX_AREAS HAL_FLASH_VERIFY_BUF_SZ MBEDTLS_AES_C MBEDTLS_BASE64_C MBEDTLS_ECP_DP_SECP224R1 MBEDTLS_ENTROPY_C MBEDTLS_PKCS1_V15 MBEDTLS_PKCS1_V21 MCU_FLASH_MIN_WRITE_SIZE MCU_STM32F4 MSYS_1_BLOCK_SIZE OS_CPUTIME_FREQ OS_CTX_SW_STACK_GUARD OS_IDLE_TICKLESS_MS_MAX OS_IDLE_TICKLESS_MS_MIN OS_MAIN_STACK_SIZE OS_MAIN_TASK_PRIO OS_SYSVIEW_TRACE_CALLOUT OS_SYSVIEW_TRACE_EVENTQ OS_SYSVIEW_TRACE_MUTEX OS_SYSVIEW_TRACE_SEM SANITY_INTERVAL STM32_HAL_I2C_HAS_CLOCKSPEED STM32_HAL_UART_HAS_SR UART_0 WATCHDOG_INTERVAL
* FLASH_OFFSET=0x8000000
* IMAGE_SLOT=0
* CORE_PATH=C:/msys64/home/mmoss/dev/myproj/repos/apache-mynewt-core
* BSP_PATH=C:/msys64/home/mmoss/dev/myproj/repos/apache-mynewt-core/hw/bsp/stm32f4discovery
* BIN_BASENAME=bin\targets\stm32f4disc_boot\app\apps\boot\boot
Error:
Downloading bin\targets\stm32f4disc_boot\app\apps\boot\boot.elf.bin to 0x8000000
Open On-Chip Debugger 0.10.0 (2018-07-28) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2-1.cfg is deprecated, please switch to interface/stlink.cfg
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v25 API v2 SWIM v14 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 2.901302
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08000188 msp: 0x10010000
auto erase enabled
Error: couldn't open bin        argetsstm32f4disc_bootppppoooot.elf.bin

load - Load application image on to the board for <target-name>

Usage:
  newt load <target-name> [flags]

Flags:
      --extrajtagcmd string   Extra commands to send to JTAG software

Global Flags:
  -h, --help              Help for newt commands
  -j, --jobs int          Number of concurrent build jobs (default 8)
  -l, --loglevel string   Log level (default "WARN")
  -o, --outfile string    Filename to tee output to
  -q, --quiet             Be quiet; only display error output
  -s, --silent            Be silent; don't output anything
  -v, --verbose           Enable verbose output when executing commands
```

This appears to be an issue with the '\\' path separator used in `$BIN_BASENAME`.  With the '\\' path separator, the `[ ! -f "$FILE_NAME" ]` in openocd.sh returns True, stopping the load process.  As a dirty work-around, I modified `common_file_to_load()` in common.sh to replace '\\' characters with '/' :

```
common_file_to_load () {
    BIN_BASENAME=$(echo $BIN_BASENAME | sed "s/\\\\/\\//g")
    if [ "$MFG_IMAGE" ]; then
        FILE_NAME=$BIN_BASENAME.bin
    elif [ "$BOOT_LOADER" ]; then
        FILE_NAME=$BIN_BASENAME.elf.bin
    else
        FILE_NAME=$BIN_BASENAME.img
    fi
}
```

With this change, the bootloader loads successfully:

```
$ newt load -v stm32f4disc_boot
Loading bootloader
Load command: C:/msys64/home/mmoss/dev/myproj/repos/apache-mynewt-core/hw/bsp/stm32f4discovery/stm32f4discovery_download.cmd C:/msys64/home/mmoss/dev/myproj/repos/apache-mynewt-core/hw/bsp/stm32f4discovery bin\targets\stm32f4disc_boot\app\apps\boot\boot
Environment:
* BOOT_LOADER=1
* FEATURES=BASELIBC_PRESENT BOOT_LOADER CONSOLE_UART_BAUD CONSOLE_UART_DEV CONSOLE_UART_FLOW_CONTROL FLASH_MAP_MAX_AREAS HAL_FLASH_VERIFY_BUF_SZ MBEDTLS_AES_C MBEDTLS_BASE64_C MBEDTLS_ECP_DP_SECP224R1 MBEDTLS_ENTROPY_C MBEDTLS_PKCS1_V15 MBEDTLS_PKCS1_V21 MCU_FLASH_MIN_WRITE_SIZE MCU_STM32F4 MSYS_1_BLOCK_SIZE OS_CPUTIME_FREQ OS_CTX_SW_STACK_GUARD OS_IDLE_TICKLESS_MS_MAX OS_IDLE_TICKLESS_MS_MIN OS_MAIN_STACK_SIZE OS_MAIN_TASK_PRIO OS_SYSVIEW_TRACE_CALLOUT OS_SYSVIEW_TRACE_EVENTQ OS_SYSVIEW_TRACE_MUTEX OS_SYSVIEW_TRACE_SEM SANITY_INTERVAL STM32_HAL_I2C_HAS_CLOCKSPEED STM32_HAL_UART_HAS_SR UART_0 WATCHDOG_INTERVAL
* FLASH_OFFSET=0x8000000
* IMAGE_SLOT=0
* CORE_PATH=C:/msys64/home/mmoss/dev/myproj/repos/apache-mynewt-core
* BSP_PATH=C:/msys64/home/mmoss/dev/myproj/repos/apache-mynewt-core/hw/bsp/stm32f4discovery
* BIN_BASENAME=bin\targets\stm32f4disc_boot\app\apps\boot\boot
Successfully loaded image.
```
The ultimate fix probably would be to use '/' path separators in `$BIN_BASENAME` when running under MinGW.

MarkMoss, 2018-11-18:
As a follow-up, it looks like TryRelPath() in util.go changes the path separators from '/' to '\\' under MinGW.

Modifying Load() in load.go so that it doesn't call TryRelPath() works around the issue for me. 

```
func (b *Builder) Load(imageSlot int, extraJtagCmd string) error {
	if b.appPkg == nil {
		return util.NewNewtError("app package not specified")
	}

	/* Populate the package list and feature sets. */
	err := b.targetBuilder.PrepBuild()
	if err != nil {
		return err
	}

	envSettings := map[string]string{
		"IMAGE_SLOT": strconv.Itoa(imageSlot),
		"FEATURES":   b.FeatureString(),
	}
	if extraJtagCmd != "" {
		envSettings["EXTRA_JTAG_CMD"] = extraJtagCmd
	}
	settings := b.cfg.SettingValues()

	var flashTargetArea string
	if parse.ValueIsTrue(settings["BOOT_LOADER"]) {
		envSettings["BOOT_LOADER"] = "1"

		flashTargetArea = "FLASH_AREA_BOOTLOADER"
		util.StatusMessage(util.VERBOSITY_DEFAULT,
			"Loading bootloader\n")
	} else {
		if imageSlot == 0 {
			flashTargetArea = "FLASH_AREA_IMAGE_0"
		} else if imageSlot == 1 {
			flashTargetArea = "FLASH_AREA_IMAGE_1"
		}
		util.StatusMessage(util.VERBOSITY_DEFAULT,
			"Loading %s image into slot %d\n", b.buildName, imageSlot+1)
	}

	bspPkg := b.targetBuilder.bspPkg
	tgtArea := bspPkg.FlashMap.Areas[flashTargetArea]
	if tgtArea.Name == "" {
		return util.NewNewtError(fmt.Sprintf("No flash target area %s\n",
			flashTargetArea))
	}
	envSettings["FLASH_OFFSET"] = "0x" + strconv.FormatInt(int64(tgtArea.Offset), 16)

	// Convert the binary path from absolute to relative.  This is required for
	// compatibility with unix-in-windows environemnts (e.g., cygwin).
	//binPath := util.TryRelPath(b.AppBinBasePath())
	binPath := b.AppBinBasePath()
	if err := Load(binPath, b.targetBuilder.bspPkg, envSettings); err != nil {
		return err
	}

	return nil
}
```

 I assume there is some environment where TryRelPath() is still needed, so removing it is probably also not a workable solution.

io-tal, 2019-04-14:
I have a similar issue on a debian machine
`Loading bootloader
Load command: /home/debian/mynewt-sr-blehci-master/repos/solidrun-mynewt-bsp/hw/bsp/nina-b1-sr_imx8m/nina-b1-sr_imx8m_download.sh /home/debian/mynewt-sr-blehci-master/repos/solidrun-mynewt-bsp/hw/bsp/nina-b1-sr_imx8m bin/targets/nina-b1_boot/app/apps/boot/boot
Environment:
* BOOT_LOADER=1
* FEATURES=BASELIBC_PRESENT BOOT_LOADER BSP_NRF52 CONSOLE_UART_BAUD CONSOLE_UART_DEV CONSOLE_UART_FLOW_CONTROL FLASH_MAP_MAX_AREAS HAL_FLASH_VERIFY_BUF_SZ I2C_0_FREQ_KHZ I2C_0_PIN_SCL I2C_0_PIN_SDA I2C_1_FREQ_KHZ MBEDTLS_AES_C MBEDTLS_BASE64_C MBEDTLS_ECP_DP_SECP224R1 MBEDTLS_ENTROPY_C MBEDTLS_PKCS1_V15 MBEDTLS_PKCS1_V21 MCU_DCDC_ENABLED MCU_FLASH_MIN_WRITE_SIZE MCU_NRF52832 MSYS_1_BLOCK_SIZE NFC_PINS_AS_GPIO OS_CPUTIME_FREQ OS_CTX_SW_STACK_GUARD OS_IDLE_TICKLESS_MS_MAX OS_IDLE_TICKLESS_MS_MIN OS_MAIN_STACK_SIZE OS_MAIN_TASK_PRIO OS_SYSVIEW_TRACE_CALLOUT OS_SYSVIEW_TRACE_EVENTQ OS_SYSVIEW_TRACE_MUTEX OS_SYSVIEW_TRACE_SEM QSPI_FLASH_SECTOR_COUNT QSPI_PIN_CS QSPI_PIN_DIO0 QSPI_PIN_DIO1 QSPI_PIN_DIO2 QSPI_PIN_DIO3 QSPI_PIN_SCK SANITY_INTERVAL SPI_0_MASTER_PIN_MISO SPI_0_MASTER_PIN_MOSI SPI_0_MASTER_PIN_SCK SPI_0_SLAVE_PIN_MISO SPI_0_SLAVE_PIN_MOSI SPI_0_SLAVE_PIN_SCK SPI_0_SLAVE_PIN_SS TIMER_0 UARTBB_0_PIN_RX UARTBB_0_PIN_TX UART_0 UART_0_PIN_CTS UART_0_PIN_RTS UART_0_PIN_RX UART_0_PIN_TX UART_1_PIN_CTS UART_1_PIN_RTS WATCHDOG_INTERVAL XTAL_32768
* FLASH_OFFSET=0x0
* IMAGE_SLOT=0
* CORE_PATH=/home/debian/mynewt-sr-blehci-master/repos/apache-mynewt-core
* BSP_PATH=/home/debian/mynewt-sr-blehci-master/repos/solidrun-mynewt-bsp/hw/bsp/nina-b1-sr_imx8m
* BIN_BASENAME=bin/targets/nina-b1_boot/app/apps/boot/boot

load - Load application image on to the board for <target-name>

Usage:
  newt load <target-name> [flags]

Flags:
      --extrajtagcmd string   Extra commands to send to JTAG software

Global Flags:
  -h, --help              Help for newt commands
  -j, --jobs int          Number of concurrent build jobs (default 1)
  -l, --loglevel string   Log level (default "WARN")
  -o, --outfile string    Filename to tee output to
  -q, --quiet             Be quiet; only display error output
  -s, --silent            Be silent; don't output anything
  -v, --verbose           Enable verbose output when executing commands
`

Could it be related? editing the common.sh file did not resolve my issue

sjanc, 2019-07-16:
Hi,

In newt 1.6 by default paths on windows are escaped. But we noticed that this behaves differently on different shells used so we also added --escape option which can be used to tune defaults eg --escape=false

could you check if this is an issue on 1.6?

ccollins476ad, 2020-06-22:
Closing this, as I believe @sjanc's suggestion solves the issue. Please reopen if there is still a problem.

closed 2020-06-22 by ccollins476ad