apache/mynewt-newt #286: Unable to build a "split target"

flyingbepi, 2019-04-16:
Trying to build a split target setup following the Mynewt documentation,
I've configured the following targets:
Loader:
_targets/thingy-loader
    app=@apache-mynewt-core/apps/bleprph
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca20020
    build_profile=optimized
    syscfg=BLE_LL_CFG_FEAT_LE_ENCRYPTION=0:BLE_SM_LEGACY=0_

Application:

 _targets/split-app
    app=@apache-mynewt-core/apps/splitty
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca20020
    build_profile=optimized
    loader=@apache-mynewt-core/apps/bleprph
    syscfg=BLE_LL_CFG_FEAT_LE_ENCRYPTION=0:BLE_SM_LEGACY=0_

The loader target builds properly but I get the following error when building the application:

_**Error: Two app packages in build: apps/bleprph, apps/splitty**_

So, either a newt tool bug or an outdated documention ?

BTW,  I didn't find any explanation in the documentation to build a monolithic (One large image; upgrade not supported) application.

Thanks.

ccollins476ad, 2019-04-26:
Hi @flyingbepi.  I am not able to replicate that exact problem, but I noticed that split image support is indeed broken.

This PR fixes the problem for me: https://github.com/apache/mynewt-newt/pull/293.  Would you mind giving it a try?

flyingbepi, 2019-04-27:
Hi Chris,

Unfortunately, this PR didn't fix my issue since I got exactly the same error message.
Regards.

PR #326 (2019-09-03): Resolve transient packages before dep resolution
This fixes #286.

If a split image specifies a transient package as the loader, newt will fail to build it.

For example, in this target:
```
target.app: "@apache-mynewt-core/apps/splitty"
target.bsp: "@apache-mynewt-core/hw/bsp/nordic_pca10040"
target.build_profile: optimized
target.loader: "@apache-mynewt-core/apps/bleprph"
```

`@apache-mynewt-core/apps/bleprph` is a transient package; it links to
`@apache-mynewt-nimble/apps/bleprph`. 

Newt reports the following error when asked to build this target:

```
Error: Two app packages in build: ap [... cut]
commit 3b5b2d66 (2019-09-03): Resolve transient packages before dep resolution

This fixes #286.

If a split image specifies a transient package as the loader, newt will
fail to build it.

For example, in this target:

    target.app: "@apache-mynewt-core/apps/splitty"
    target.bsp: "@apache-mynewt-core/hw/bsp/nordic_pca10040"
    target.build_profile: optimized
    target.loader: "@apache-mynewt-core/apps/bleprph"

`@apache-mynewt-core/apps/bleprph` is a transient package; it links to
`@apache-mynewt-nimble/apps/bleprph`.

Newt reports the following error when asked to build this target:

    Error: Two app packages in  [... cut]
closed 2019-09-03 by ccollins476ad