apache/mynewt-newt #297: "Invalid version string" error from 'newt install' is borederline useless

bjc, 2019-05-14:
I have no idea what the problem is (except that, somehow, "master" is an invalid version), nor where the problem is emanating from.

At a minimum, it should show which repository is causing the problem. Ideally it should also show how it derived that version and from which `vers` keys it was obtained.

ccollins476ad, 2019-05-14:
Almost as useless as this bug report :)

Could you please include:
* output of `newt version`
* output of `newt info`
* your `project.yml` file
* transcript of the session where this error was produced

Thanks,
Chris

bjc, 2019-05-14:
The problem exists in master, and starts from here: https://github.com/apache/mynewt-newt/blob/4f0c27e12d030f7de917846f2afc735d87be2359/newt/newtutil/repo_version.go#L194

Problematic repository is here: https://github.com/makerdiary/mynewt_nrf52840_mdk

ccollins476ad, 2019-05-15:
Thanks @bjc.  I have submitted this PR to improve this error message: https://github.com/apache/mynewt-newt/pull/298

You may have already figured this out, but here is an explanation of the problem for completeness.  Newt complains about that `mynewt_nrf52840_mdk` repo because of this line in its `repository.yml` file:
```
    "0-latest": "master"
```

A "floating version" (e.g., `0-latest`, `1.0-stable`) can only point to a fixed version number (e.g., `1.4.0`).  Each fixed version number then points to a git commit (branch, hash, or tag).  A floating version is not allowed to point directly to a git commit.

I am not sure if this is documented anywhere.

PR #298 (2019-05-15): Improve errors for invalid repo version strings
This fixes #297.
commit 60fc2df0 (2019-05-15): Improve errors for invalid repo version strings

This fixes #297.
closed 2019-05-15 by ccollins476ad