apache/mynewt-newt #310: build.sh on windows is not creating binary where it should

sjanc, 2019-07-17:
newt binary is left in /tmp folder which deleted at then end of script...

There seems to be some subtle change in go 1.12 about how it handles paths..

utzig, 2019-07-17:
If just run `go build` on newt path does it work?

sjanc, 2019-09-25:
yeap, it works just fine and binary is in folder I executed 'go build'.
I suspect build.sh magic with tmp godir etc is causing issues...

sjanc, 2020-03-26:
ok, the issue was not running msys2 shell as administrator (which prevented symlinks from working). Closing.

closed 2020-03-26 by sjanc

dsilva-vd, 2020-09-28:
Hi @sjanc,

Running the msys2 shell as administrator didn't work for me on Windows 10 system. 

For those trying to build newt from source before running ./build.sh open the file and comment out rm -rf "$godir"

Run ./build.sh
Go to your msys2 install directory and look inside the /tmp folder. You will find the newt build directory and then navigate to /newt and you will find the newt.exe file.