apache/mynewt-newt #376: newt upgrade throws error

Abhiroop, 2020-02-13:
Running `newt upgrade -v` results in the following error:

```
Error: Error updating "mcuboot": error: The following untracked working tree files would be overwritten by checkout:
	ext/mbedtls/include/mbedtls/asn1.h
	ext/mbedtls/include/mbedtls/bignum.h
	ext/mbedtls/include/mbedtls/check_config.h
	ext/mbedtls/include/mbedtls/config.h
	ext/mbedtls/include/mbedtls/ecdsa.h
	ext/mbedtls/include/mbedtls/ecp.h
	ext/mbedtls/include/mbedtls/md.h
	ext/mbedtls/include/mbedtls/oid.h
	ext/mbedtls/include/mbedtls/pk.h
	ext/mbedtls/include/mbedtls/platform.h
	ext/mbedtls/include/mbedtls/platform_util.h
	ext/mbedtls/include/mbedtls/threading.h
Please move or remove them before you switch branches.
Aborting
```

I am following the create project documentation here : https://mynewt.apache.org/latest/get_started/project_create.html

I am running an Ubuntu 18.04 natively in a MacBook Pro 13''

utzig, 2020-02-13:
This was fixed some time ago, and will be available on next release. If you installed through the Ubuntu packages, you can try uninstalling the package, and then install using the go tooling: `go get -v mynewt.apache.org/newt/newt`. This should install `newt` under `$GOPATH/bin`.

Abhiroop, 2020-02-21:
Thanks it works with `go get`.

closed 2020-02-21 by Abhiroop