apache/mynewt-newt #398: Not able to build newt example project on OS X

NilsMinor, 2020-05-04:
Hey newt-team,

I am trying to run the example project https://mynewt.apache.org/v1_5_0/tutorials/blinky/arduino_zero.html but I have some problems building it.

I followed the instructions from the arduino_zero tutorial.
After creating the target I run newt upgrade I get the following versions.

  ```
  upgrade apache-mynewt-core (1.7.0 --> 1.8.0)
    upgrade apache-mynewt-mcumgr (0.0.0 --> 0.1.0)
    upgrade apache-mynewt-nimble (1.2.0 --> 1.3.0)
    upgrade mcuboot (0.0.0 --> 1.5.0)
    install mynewt_arduino_zero (1.8.0)
```

my project.yml looks like :
```
project.repositories:
    - apache-mynewt-core
    - mynewt_arduino_zero

# Use github's distribution mechanism for core ASF libraries.
# This provides mirroring automatically for us.
#
repository.apache-mynewt-core:
    type: github
    vers: 1-latest
    user: apache
    repo: mynewt-core
```

repository.mynewt_arduino_zero:
    type: github
    vers: 1-latest
    user: runtimeco
    repo: mynewt_arduino_zero

But I am always failing to build the target. The output looks like:
```
Error: repos/apache-mynewt-core/kernel/os/src/os_mempool.c: In function 'os_memblock_from':
repos/apache-mynewt-core/kernel/os/src/os_mempool.c:308:5: error: implicit declaration of function 'static_assert' [-Werror=implicit-function-declaration]
     static_assert(sizeof block_addr == sizeof baddr32,
     ^
cc1: all warnings being treated as errors
```

I am using OS X (10.15.4 Catalina) with the following gcc (gcc --version)

```
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.3 (clang-1103.0.32.59)
Target: x86_64-apple-darwin19.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

Newt version:
```
Apache Newt
   Version: 1.7.0
  Git Hash: 80bcba7
Build Date: 2019-08-06_13:20
```

I tested different newt versions (latest 1.7, 1.8 ..) and I changed the git versions in the project.yml. But I was not able to build the example. What am I doing wrong? Any help would be appreciated :)

BTW, newt looks really like an awesome project!

Thanks,
Nils

ccollins476ad, 2020-05-04:
Hi @NilsMinor ,

It looks like your version of gcc is using an old C standard by default (older than C11).  This can be fixed by upgrading your version of gcc.  Note: newt does not use `gcc` when building that target, it uses `arm-none-eabi-gcc` (cross compiler).  You can check your gcc version with
```
arm-none-eabi-gcc -v
```

The recommended minimum gcc version for Mynewt is gcc 7.

NilsMinor, 2020-05-05:
Hi @ccollins476ad ,

thanks for your help. You were right, I used the wrong compiler + an old version ^^.

I updated my ar-none-eabi-gcc. 

arm-none-eabi-gcc -v
```
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/lto-wrapper
Target: arm-none-eabi
Configured with: /tmp/jenkins-GCC-9-pipeline-100_20191030_1572397548/src/gcc/configure --target=arm-none-eabi --prefix=/tmp/jenkins-GCC-9-pipeline-100_20191030_1572397548/install-native --libexecdir=/tmp/jenkins-GCC-9-pipeline-100_20191030_1572397548/install-native/lib --infodir=/tmp/jenkins-GCC-9-pipeline-100_20191030_1572397548/install-native/share/doc/gcc-arm-none-eabi/info --mandir=/tmp/jenkins-GCC-9-pipeline-100_20191030_1572397548/install-native/share/doc/gcc-arm-none-eabi/man --htmldir=/tmp/jenkins-GCC-9-pipeline-100_20191030_1572397548/install-native/share/doc/gcc-arm-none-eabi/html --pdfdir=/tmp/jenkins-GCC-9-pipeline-100_20191030_1572397548/install-native/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/tmp/jenkins-GCC-9-pipeline-100_20191030_1572397548/install-native/arm-none-eabi --build=x86_64-apple-darwin10 --host=x86_64-apple-darwin10 --with-gmp=/tmp/jenkins-GCC-9-pipeline-100_20191030_1572397548/build-native/host-libs/usr --with-mpfr=/tmp/jenkins-GCC-9-pipeline-100_20191030_1572397548/build-native/host-libs/usr --with-mpc=/tmp/jenkins-GCC-9-pipeline-100_20191030_1572397548/build-native/host-libs/usr --with-isl=/tmp/jenkins-GCC-9-pipeline-100_20191030_1572397548/build-native/host-libs/usr --with-libelf=/tmp/jenkins-GCC-9-pipeline-100_20191030_1572397548/build-native/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-lstdc++ -lm' --with-pkgversion='GNU Tools for Arm Embedded Processors 9-2019-q4-major' --with-multilib-list=rmprofile
Thread model: single
gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (GNU Tools for Arm Embedded Processors 9-2019-q4-major)
```
I use now the latest version 9.2.1

I changed the same project to 1-latest and 0-dev and used newt 1.9.0-dev and 1.7.0 but when building the app I now get this error. What have I missed?

```
Error: In file included from repos/apache-mynewt-core/kernel/os/include/os/os_fault.h:26,
                 from repos/apache-mynewt-core/libc/baselibc/include/assert.h:24,
                 from repos/mcuboot/boot/mynewt/src/main.c:22:
repos/mcuboot/boot/mynewt/src/main.c: In function 'main':
repos/apache-mynewt-core/hw/hal/include/hal/hal_debug.h:49:17: error: implicit declaration of function 'hal_debug_break' [-Werror=implicit-function-declaration]
   49 |                (hal_debug_break(), 1))
      |                 ^~~~~~~~~~~~~~~
repos/apache-mynewt-core/kernel/os/include/os/os_fault.h:38:21: note: in expansion of macro 'HAL_DEBUG_BREAK'
   38 | #define OS_CRASH() (HAL_DEBUG_BREAK(), __assert_func(NULL, 0, NULL, NULL))
      |                     ^~~~~~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/include/assert.h:26:36: note: in expansion of macro 'OS_CRASH'
   26 | #define assert(x) ((x) ? (void)0 : OS_CRASH())
      |                                    ^~~~~~~~
repos/mcuboot/boot/mynewt/src/main.c:222:5: note: in expansion of macro 'assert'
  222 |     assert(rc == 0);
      |     ^~~~~~
cc1: all warnings being treated as errors
```

Thanks for your support,
Nils

closed 2020-05-06 by NilsMinor