apache/mynewt-newt #399: Question on how to follow mynewt development process the right way.

NilsMinor, 2020-05-06:
Hi,

I want to use Mynewt and nimble an a project using a custom created board powered by an NRF52832 chip. I have some questions in order to follow the your development process.

I started testing Mynewt by creating an example project for the nrf52 DK. The core and nimble repos are downloaded to my app. I have some questions on carry on.

- What is the correct way to create a bsp for my own board? I think I should copy an example bsp and change the values to fit my board (add pwm, set io's ...), correct? 
- How to maintain my project? newt downloads the a specific version to my repo, should I apply changes to this repo? I normally fork projects form Github, crate my own branch and apply changes to it in order to update to latest master when required. But I do not really understand how to correctly develop code for newt/nimble.

Sorry for the nooby questions, I hope I did not miss anything in the documentations but I want to follow the development flow correctly.

Best regards,
Nils

ccollins476ad, 2020-05-06:
> What is the correct way to create a bsp for my own board? I think I should copy an example bsp and change the values to fit my board (add pwm, set io's ...), correct?

You might find this helpful: http://mynewt.apache.org/latest/os/core_os/porting/port_bsp.html.  Copying an existing BSP is also a reasonable way to go about doing this.

> How to maintain my project? newt downloads the a specific version to my repo, should I apply changes to this repo? I normally fork projects form Github, crate my own branch and apply changes to it [...]

Creating a fork+branch is a good way to do it.  Then modify your project.yml file so that it specifies your fork+branch.  For example:

```
repository.apache-mynewt-nimble:
    type: git
    vers: '<branch-name>-commit'
    url: "git@github.com:NilsMinor/mynewt-nimble.git"
```

When you run `newt upgrade`, newt will checkout the specified branch from your nimble fork.

NilsMinor, 2020-05-07:
Thanks @ccollins476ad this fixed my knowledge gap!

closed 2020-05-07 by NilsMinor