apache/mynewt-newt #425: newt test does not work with Mac OS X Catalina

vrahane, 2020-12-09:
```
ld: warning: The i386 architecture is deprecated for macOS (remove from the Xcode build setting: ARCHS)
ld: warning: ignoring file /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/lib/libm.tbd, missing required architecture i386 in file /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/lib/libm.tbd
ld: warning: ignoring file /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/lib/libSystem.tbd, missing required architecture i386 in file /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/lib/libSystem.tbd
Undefined symbols for architecture i386:
  "_cfmakeraw", referenced from:
      _uart_dev_set_attr in @apache-mynewt-core_hw_mcu_native.a(native_uart_cfg.o)
  "_cfsetispeed", referenced from:
      _uart_dev_set_attr in @apache-mynewt-core_hw_mcu_native.a(native_uart_cfg.o)
  "_isalnum", referenced from:
      _uart_log_data in @apache-mynewt-core_hw_mcu_native.a(hal_uart.o)
  "___snprintf_chk", referenced from:
      _uart_pty in @apache-mynewt-core_hw_mcu_native.a(hal_uart.o)
      _uart_log_data in @apache-mynewt-core_hw_mcu_native.a(hal_uart.o)
  "_isdigit", referenced from:
      _uart_hal_init in @apache-mynewt-core_hw_drivers_uart_uart_hal.a(uart_hal.o)
  "_sigaction", referenced from:
      _sim_signals_init in @apache-mynewt-core_kernel_sim.a(sim_sched_sig.o)
  "_setitimer", referenced from:
      _sim_tick_idle in @apache-mynewt-core_kernel_sim.a(sim_sched_sig.o)
  "_tcgetattr", referenced from:
      _uart_pty_set_attr in @apache-mynewt-core_hw_mcu_native.a(hal_uart.o)
  "_sigprocmask", referenced from:
      _sim_save_sr in @apache-mynewt-core_kernel_sim.a(sim_sched_sig.o)
      _sim_restore_sr in @apache-mynewt-core_kernel_sim.a(sim_sched_sig.o)
      _sim_in_critical in @apache-mynewt-core_kernel_sim.a(sim_sched_sig.o)
  "_memcmp", referenced from:
      _addrcmp in @apache-mynewt-core_net_ip_native_sockets.a(native_itf.o)
  "_if_nametoindex", referenced from:
      _native_sock_itf_getnext in @apache-mynewt-core_net_ip_native_sockets.a(native_itf.o)
      _native_sock_itf_addr in @apache-mynewt-core_net_ip_native_sockets.a(native_itf.o)
      _native_sock_itf_addr_getnext in @apache-mynewt-core_net_ip_native_sockets.a(native_itf.o)
  "_freeifaddrs", referenced from:
      _native_sock_itf_getnext in @apache-mynewt-core_net_ip_native_sockets.a(native_itf.o)
      _native_sock_itf_addr in @apache-mynewt-core_net_ip_native_sockets.a(native_itf.o)
      _native_sock_itf_addr_getnext in @apache-mynewt-core_net_ip_native_sockets.a(native_itf.o)
  "_open$UNIX2003", referenced from:
      _flash_native_file_open in @apache-mynewt-core_hw_mcu_native.a(hal_flash.o)
      _uart_open_dev in @apache-mynewt-core_hw_mcu_native.a(hal_uart.o)
      _uart_open_log in @apache-mynewt-core_hw_mcu_native.a(hal_uart.o)
  "_kill$UNIX2003", referenced from:
```

vrahane, 2020-12-09:
cc @ccollins476ad @nkaje @aditihilbert

sjanc, 2026-10-31:
this is fixed in latest mynewt; some networking tests are failing but this is not due to newt tool

closed 2026-10-31 by sjanc