apache/mynewt-newt #447: Downloading repos fail

danielkucera, 2021-10-10:
Hi,
I am not sure if I do something wrong but getting nimble repo fails when I empty the `repos` directory:
running:
```
rm -rf repos/*
newt upgrade --verbose
```
result:
```
Downloading repository mynewt-core (commit: master) from https://github.com/apache/mynewt-core.git
Cloning into '/tmp/newt-repo463302550'...
remote: Enumerating objects: 123950, done.
remote: Counting objects: 100% (2632/2632), done.
remote: Compressing objects: 100% (1178/1178), done.
remote: Total 123950 (delta 1344), reused 2393 (delta 1255), pack-reused 121318
Receiving objects: 100% (123950/123950), 172.20 MiB | 4.62 MiB/s, done.
Resolving deltas: 100% (75945/75945), done.
Updating files: 100% (6807/6807), done.
Will checkout d8737c05cdfb1f0615962e13feb2cf7968c70dee
Downloading repository mynewt-nimble (commit: master) from https://github.com/apache/mynewt-nimble.git
Cloning into '/tmp/newt-repo133076989'...
remote: Enumerating objects: 41918, done.
remote: Counting objects: 100% (840/840), done.
remote: Compressing objects: 100% (555/555), done.
remote: Total 41918 (delta 426), reused 534 (delta 252), pack-reused 41078
Receiving objects: 100% (41918/41918), 12.87 MiB | 3.74 MiB/s, done.
Resolving deltas: 100% (25525/25525), done.
Will checkout fbd0a8c24d369a17dc04281f58a17ed7392e7f22
[apache-mynewt-nimble]:
Downloading repository description
Fetching repo mynewt-nimble
Download of "repository.yml" from repo:apache-mynewt-nimble commit:master successful
Downloading repository mcuboot (commit: master) from https://github.com/mcu-tools/mcuboot.git
Cloning into '/tmp/newt-repo180196152'...
remote: Enumerating objects: 12592, done.
remote: Counting objects: 100% (1576/1576), done.
remote: Compressing objects: 100% (706/706), done.
remote: Total 12592 (delta 923), reused 1194 (delta 803), pack-reused 11016
Receiving objects: 100% (12592/12592), 4.49 MiB | 4.24 MiB/s, done.
Resolving deltas: 100% (8028/8028), done.
Will checkout 137d79717764ed32d5da4b4b301f32f81b2bf40f
Skipping submodule updates
[mcuboot]:
Downloading repository description
Fetching repo mcuboot
Download of "repository.yml" from repo:mcuboot commit:master successful
Downloading repository mynewt-mcumgr (commit: master) from https://github.com/apache/mynewt-mcumgr.git
Cloning into '/tmp/newt-repo720763703'...
remote: Enumerating objects: 2532, done.
remote: Counting objects: 100% (253/253), done.
remote: Compressing objects: 100% (168/168), done.
remote: Total 2532 (delta 127), reused 136 (delta 58), pack-reused 2279
Receiving objects: 100% (2532/2532), 594.26 KiB | 4.91 MiB/s, done.
Resolving deltas: 100% (1239/1239), done.
Will checkout 12b496e37caf20a45ab5aee4209b06c5d79ef9b1
[apache-mynewt-mcumgr]:
Downloading repository description
Fetching repo mynewt-mcumgr
Download of "repository.yml" from repo:apache-mynewt-mcumgr commit:master successful
* Warning: Multiple packages with same pkg.name=targets/unittest in repo my_project; path1=/github/workspace/targets/unittest path2=/github/workspace/tmp/apache-mynewt-core/targets/unittest
* Warning: Multiple packages with same pkg.name=targets/unittest in repo my_project; path1=/github/workspace/targets/unittest path2=/github/workspace/tmp/apache-mynewt-nimble/targets/unittest
* Warning: Multiple packages with same pkg.name=apps/blinky in repo my_project; path1=/github/workspace/apps/blinky path2=/github/workspace/tmp/mcuboot/testplan/mynewt/apps/blinky
* Warning: Multiple packages with same pkg.name=apps/slinky in repo my_project; path1=/github/workspace/tmp/apache-mynewt-core/apps/slinky path2=/github/workspace/tmp/mcuboot/testplan/mynewt/apps/slinky
[apache-mynewt-core]:
Downloading repository description
Fetching repo mynewt-core
Download of "repository.yml" from repo:apache-mynewt-core commit:master successful
Making the following changes to the project:
Will checkout b9ca5270c7954815c41e4451345f04d300b8e9fa
apache-mynewt-core successfully upgraded to version 1.9.0
[... 1634 lines cut ...]
+#endif
+
+#ifndef MYNEWT_VAL_BSP_nordic_pca10056
+#define MYNEWT_VAL_BSP_nordic_pca10056 (1)
+#endif
+
+#ifndef MYNEWT_VAL_NEWT_FEATURE_LOGCFG
+#define MYNEWT_VAL_NEWT_FEATURE_LOGCFG (1)
+#endif
+
+#ifndef MYNEWT_VAL_NEWT_FEATURE_SYSDOWN
+#define MYNEWT_VAL_NEWT_FEATURE_SYSDOWN (1)
+#endif
+
+#ifndef MYNEWT_VAL_TARGET_NAME
+#define MYNEWT_VAL_TARGET_NAME ("riot")
+#endif
+
+#ifndef MYNEWT_VAL_TARGET_riot
+#define MYNEWT_VAL_TARGET_riot (1)
+#endif
+
+#endif
```
How do I fix this?

sjanc, 2022-03-14:
HI, sorry I miss comment about minor release.  We will make 1.10 release soon so hope using master branch wasn't a big issue for you

danielkucera, 2022-03-14:
No worries, you can close this.

closed 2022-04-15 by sjanc