apache/mynewt-newt #494: Not able to build blinky with newt 1.9.0 and newt 1.10.0

uwej711, 2022-11-13:
Hi, 

I wanted to give mynewt a try and installed newt 1.10.0 on a raspberry pi. I created a new project with newt new --shallow=0 blinky and added a target. When I run newt build my_target it goes directly to linking without compiling any files which results in a warning since Reset_Handler is not there. With newt 1.9.0 it is the same. But with newt 1.8.0 all the files are compiled and there is no linker warning.
It looks like the packages/dependencies for blinky are not properly resolved. 
Any help would be appreciated.

Kind regards
Uwe

sjanc, 2022-11-14:
Hi,

I just tested it briefly on linux PC with newt 1.10 and all seem to work OK.   Didn't run it on PI though...  
Could you add some logs on how you setup your project?

uwej711, 2022-11-14:
Here is what I use: newt build from source

[alarm@alarm ~]$ go version
go version go1.19.3 linux/arm64

[alarm@alarm ~]$ newt version
Apache Newt 1.10.0 / 7b428d5-dirty / 2022-11-14_19:22

[alarm@alarm mynewt]$ newt new --shallow 0 blinky_10
Downloading project skeleton from apache/mynewt-blinky...
Downloading repository mynewt-blinky (commit: master) from https://github.com/apache/mynewt-blinky.git
Installing skeleton in blinky_10 (commit: mynewt_1_10_0_rc1_tag)
Project blinky_10 successfully created.

[alarm@alarm mynewt]$ cd blinky_10/
[alarm@alarm blinky_10]$ newt upgrade
Downloading repository mynewt-core (commit: master) from https://github.com/apache/mynewt-core.git
WARNING: Creating a shallow clone by default. To unshallow repository later use "git fetch --unshallow".
Downloading repository mynewt-mcumgr (commit: master) from https://github.com/apache/mynewt-mcumgr.git
Downloading repository mynewt-nimble (commit: master) from https://github.com/apache/mynewt-nimble.git
Downloading repository mcuboot (commit: main) from https://github.com/mcu-tools/mcuboot.git
Making the following changes to the project:
apache-mynewt-core successfully upgraded to version 1.10.0
apache-mynewt-mcumgr successfully upgraded to version 0.3.0
apache-mynewt-nimble successfully upgraded to version 1.5.0
mcuboot successfully upgraded to version 1.7.2

Create a target for the bootloader ...

[alarm@alarm blinky_10]$ newt target show
targets/ble400_boot
    app=@mcuboot/boot/mynewt
    bsp=@apache-mynewt-core/hw/bsp/ble400
    build_profile=optimized

I can include the complete output from ```newt build -ldebug ble400_boot``` but it is quite long, maybe just let my know what would be of interest.

uwej711, 2022-11-14:
In comparision, with newt 1.8.0 newt starts to compile files after:

2022/11/14 19:51:18.228 [DEBUG] Generating build flags for package @apache-mynewt-core/util/rwlock
2022/11/14 19:51:18.228 [DEBUG] Generating build flags for compiler
2022/11/14 19:51:18.230 [DEBUG] arm-none-eabi-gcc -DAPP_NAME=mynewt 

with newt 1.10.0 no compiling but directly:

2022/11/14 19:38:14.701 [DEBUG] Generating build flags for package @apache-mynewt-core/util/rwlock
2022/11/14 19:38:14.701 [DEBUG] Generating build flags for compiler
2022/11/14 19:38:14.703 [DEBUG] created user pre-link dir: /tmp/mynewt-user-pre-link1932577823

closed 2022-11-21 by sjanc