apache/mynewt-newt #567: newt mfg create panic

jfclere, 2024-09-04:
I have:

``
jfclere@fedora:~/dev/try$ newt mfg create rb_blinky_rsa 0.0.1
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x603f36]

goroutine 1 [running]:
mynewt.apache.org/newt/newt/pkg.(*BspPackage).getAutogeneratedLinkerScriptPath(0x0?)
	/tmp/mynewt.2vigQrJxDF/src/mynewt.apache.org/newt/newt/pkg/bsp_package.go:86 +0x16
mynewt.apache.org/newt/newt/pkg.(*BspPackage).resolveLinkerScriptSetting(0xc0009db320, 0xc00061b030, {0x7ce6f5, 0x10})
	/tmp/mynewt.2vigQrJxDF/src/mynewt.apache.org/newt/newt/pkg/bsp_package.go:139 +0x27b
mynewt.apache.org/newt/newt/pkg.(*BspPackage).Reload(0xc0009db320, 0x5?)
	/tmp/mynewt.2vigQrJxDF/src/mynewt.apache.org/newt/newt/pkg/bsp_package.go:206 +0x46f
mynewt.apache.org/newt/newt/pkg.NewBspPackage(0xc0001a9980, 0x0)
	/tmp/mynewt.2vigQrJxDF/src/mynewt.apache.org/newt/newt/pkg/bsp_package.go:265 +0x165
mynewt.apache.org/newt/newt/mfg.calcBsp({{0xc000128c00, 0x2, 0x2}, {0x0, 0x0, 0x0}, 0xc0006c4a50, {0xc000a351d0, 0x2a}}, 0xc00020c600)
	/tmp/mynewt.2vigQrJxDF/src/mynewt.apache.org/newt/newt/mfg/build.go:166 +0x1fd
mynewt.apache.org/newt/newt/mfg.newMfgBuilder(0xc000c17840?, {{0xc000128c00, 0x2, 0x2}, {0x0, 0x0, 0x0}, 0xc0006c4a50, {0xc000a351d0, 0x2a}}, ...)
	/tmp/mynewt.2vigQrJxDF/src/mynewt.apache.org/newt/newt/mfg/build.go:432 +0xb7
mynewt.apache.org/newt/newt/mfg.LoadMfgEmitter(_, {_, _, _, _}, {_, _, _}, _)
	/tmp/mynewt.2vigQrJxDF/src/mynewt.apache.org/newt/newt/mfg/misc.go:54 +0x1df
mynewt.apache.org/newt/newt/cli.mfgCreateRunCmd(0xc0001fc608, {0xc00013cb60, 0x2, 0x2})
	/tmp/mynewt.2vigQrJxDF/src/mynewt.apache.org/newt/newt/cli/mfg_cmds.go:119 +0x20f
github.com/spf13/cobra.(*Command).execute(0xc0001fc608, {0xc00013cb00, 0x2, 0x2})
	/home/jfclere/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:920 +0x867
github.com/spf13/cobra.(*Command).ExecuteC(0xc0001cc008)
	/home/jfclere/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:1044 +0x3a5
github.com/spf13/cobra.(*Command).Execute(...)
	/home/jfclere/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:968
main.main()
	/tmp/mynewt.2vigQrJxDF/src/mynewt.apache.org/newt/newt/newt.go:184 +0x25c
``

I have 2 projects:
``
jfclere@fedora:~/dev/try$ newt target show DK_blinky
targets/DK_blinky
    app=apps/blinky
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca10056
``
and:
``
jfclere@fedora:~/dev/try$ newt target show nrf52_boot
targets/nrf52_boot
    app=@mcuboot/boot/mynewt
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca10056
    build_profile=optimized
``

for configuration:
``
jfclere@fedora:~/dev/try$ more  mfgs/rb_blinky_rsa/mfg.yml
mfg.bsp: "@apache-mynewt-core/hw/bsp/nordic_pca10056"
mfg.targets:
    - rb_boot:
      name: "targets/nrf52_boot"
      area: FLASH_AREA_BOOTLOADER
      offset: 0x0
    - rb_blinky:
      name: "targets/DK_blinky"
      area: FLASH_AREA_IMAGE_0
      offset: 0x0
mfg.meta:
    area: FLASH_AREA_BOOTLOADER
``
and:
``
jfclere@fedora:~/dev/try$ more mfgs/rb_blinky_rsa/pkg.yml
pkg.name: "mfgs/rb_blinky_rsa"
pkg.type: "mfg"
pkg.description:
pkg.author:
pkg.homepage:
``
To get the panic:
``
newt  build nrf52_boot
newt  build DK_blinky
newt create-image DK_blinky 0.0.1
newt mfg create rb_blinky_rsa 0.0.1
``

m-gorecki, 2024-09-06:
Hi, thanks for reaching out. As for the panic error, I've created the PR with fix: https://github.com/apache/mynewt-newt/pull/568

Now manufacturing image should be created for your targets, but I do still encounter issues with newt mfg load command. I'll investigate this further next week. However flashing created manufacturing image with JLinkGDBServer and GDB or with nrfjprog works fine.

jfclere, 2024-09-07:
Thanks the PR fixes my problem.

closed 2024-09-08 by m-gorecki